wzpan/dingdang-robot #4: 手动安装遇到的问题

wwj718, 2017-06-11:
Hi wzpan，

我想在本地手动安装环境，stt_passive_engine想使用snowboy

wiki的安装文档里不知能否补上这个

wzpan, 2017-06-11:
snowboy 不需要额外安装什么东西呢，直接把 passive stt 引擎改成 snowboy 即可。

wwj718, 2017-06-11:
@wzpan  我这边报了这个错呢
[image]

wzpan, 2017-06-11:
```
stt_passive_engine: snowboy-stt
```

wwj718, 2017-06-11:
感谢! 这个问题解决了，目前遇到 `ValueError: TTS engine 'baidu-tts' is not available (due to missing dependencies, etc.)`

我已经安装baidu-tts了

wzpan, 2017-06-11:
得看前面的log。missing 了什么 dependencies 。估计是少安装了 pydub 之类的库。

wwj718, 2017-06-11:
可能是系统依赖问题，想在mac下搭建开发环境，一些linux的库可能没有，之后还是切换到树莓派里去好了

wzpan, 2017-06-11:
可能不兼容Mac。

wwj718, 2017-06-11:
嗯，还是在树莓派折腾好啦，环境问题太烦人

另错误信息里没提示缺少什么依赖

昨天在群里看到你这边呼吁大家做的几个插件，有一些之前我试着做过，想加上。一个是nlp相关的插件(定时任务) 另一个是语音控制电脑 , 博客里有折腾这块的玩具： http://blog.just4fun.site/pi-guardian.html

wzpan, 2017-06-11:
好，欢迎你的pull request！
语音控制电脑可能需要思考下有没有比较通用的协议，不仅仅是启动应用这么简单，最好还能做到控制某个应用（比如浏览器）。Echo就可以控制浏览网页。

wwj718, 2017-06-11:
控制浏览器的话，python有驱动的，可以很精细地控制。这块是否需要整体做个规划，定义出intent呢

wzpan, 2017-06-11:
还是需要的，方便别人也能扩展最好。

wwj718, 2017-06-11:
👌

closed 2017-06-11 by wwj718