wzpan/dingdang-robot #11: 镜像中系统源替换不彻底

GoldJohnKing, 2017-08-03:
只替换了raspbian源，没有替换raspi源（这个源貌似是Debian系统已经抛弃但是Raspbian还在使用的软件包的源，据观察，阿里云上没有这一目录，只有ustc有……）：
Quote:
修改/etc/apt/sources.list.d/raspi.list为以下内容：
deb https://mirrors.ustc.edu.cn/archive.raspberrypi.org/ jessie main ui

（经个人测试……感觉ustc的速度比阿里云快……）
Quote:
编辑/etc/apt/sources.list文件。删除原文件所有内容，用以下内容取代：
deb http://mirrors.ustc.edu.cn/raspbian/raspbian/ jessie main non-free contrib
deb-src http://mirrors.ustc.edu.cn/raspbian/raspbian/ jessie main non-free contrib

GoldJohnKing, 2017-08-03:
还有一件事……镜像系统提供的resource.list的第二个源的最后少了个"/"，本强迫症建议有机会加上……

wzpan, 2017-08-11:
你的个人测试未必能代表所有人的速度：

``` sh
pi@raspberrypi:~/dingdang $ sudo apt-get update
忽略 https://mirrors.ustc.edu.cn jessie InRelease      
错误 http://mirrors.ustc.edu.cn jessie InRelease        
  
错误 http://mirrors.ustc.edu.cn jessie Release.gpg      
  无法发起与 mirrors.ustc.edu.cn:80 (2001:da8:d800:95::110) 的连接 - connect (101: 网络不可达) [IP: 2001:da8:d800:95::110 80]

```

GoldJohnKing, 2017-08-12:
你可能是网络环境有问题……

wzpan, 2017-08-13:
新版镜像即将发布，加了 raspi.list 的镜像源。其他的保持阿里云镜像不变。补上了 `/` 。

closed 2017-08-13 by GoldJohnKing