wzpan/dingdang-robot #12: 考虑到rc.local以root用户身份执行的权限问题，建议修改开机启动教程

GoldJohnKing, 2017-08-04:
采用如下方法，可以免去建立软链接，同时解决rc.local以root用户身份执行的权限问题（如网易云音乐因缺乏写权限而无法使用）。

在 /home/pi/.config 下找到autostart的文件夹。在该文件夹下创建一个空文件，文件名自拟，后缀必须是desktop，如：dingdang.desktop。
用记事本打开该文件，修改文件内容如下并保存：

```
[Desktop Entry]
Name=Dingdang
Comment=Dingdang Robot
Exec=sudo python /home/pi/dingdang/dingdang.py
Icon=/home/pi/python_games/4row_black.png
Terminal=false
MultipleArgs=false
Type=Application
Categories=Application;Development;
StartupNotify=true
```

Name、Comment、Icon 可以自定，分别表示启动项的名称、备注和图标。Exec 表示调用的指令，和在终端输入运行脚本的指令格式一致。为了使叮当的重新启动插件可用（重新启动插件要求root权限），在Exec部分增加了sudo。如果希望叮当在开机时在终端中启动，可以修改exec为：`sudo lxterminal -e "python /home/pi/dingdang/dingdang.py"`

重新启动系统后将会自动以默认用户（pi）身份启动叮当。

（但是这样貌似就没有办法 sleep 5 了，在本人的十几次测试中，有两次发生叮当无法正常播放语音。如果有办法sleep，最好是想办法加上。个人考虑可以在rc.local中sleep，在这个里面启动。但是不知道rc.local的sleep是否能有效改善这一情况，还有待观察和测试。）

参考链接：http://blog.csdn.net/A_lPha/article/details/53515943

GoldJohnKing, 2017-08-04:
确认修改之前希望能够确确实实地验证一下，尤其是是否可以省略软链接。本人在本机测试成功，但不排除删除软链接后导致的其他可能存在的潜在问题。

wzpan, 2017-08-04:
好，明天试试。

wzpan, 2017-08-06:
可以把 Exec 改为执行本地的一个 shell 脚本，在里头写 `sleep 5` 。

wzpan, 2017-08-06:
已经改好：<https://github.com/wzpan/dingdang-robot/wiki/configuration#设置开机启动>

closed 2017-08-06 by GoldJohnKing