wzpan/dingdang-robot #13: 启动后报错

rekcah1986, 2017-08-04:
[INFO] Please use WeChat to scan the QR code .
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, self->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2719
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
Traceback (most recent call last):
  File "./dingdang.py", line 188, in <module>
    app.run()
  File "./dingdang.py", line 154, in run
    conversation.handleForever()
  File "/home/pi/dingdang/client/conversation.py", line 63, in handleForever
    threshold, transcribed = self.mic.passiveListen(self.persona)
  File "/home/pi/dingdang/client/mic.py", line 140, in passiveListen
    frames_per_buffer=CHUNK)
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 747, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 442, in __init__
    self._stream = pa.open(**arguments)
IOError: [Errno Invalid sample rate] -9997
[INFO] Please confirm to login .
[INFO] Web WeChat login succeed .
[INFO] Web WeChat init succeed .
[INFO] Get 155 contacts
[INFO] Start to process messages .
ERROR:apscheduler.executors.default:Job "Notifier.gather (trigger: interval[0:00:30], next run at: 2017-08-05 06:23:20 HKT)" raised an exception
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/apscheduler/executors/base.py", line 108, in run_job
    retval = job.func(*job.args, **job.kwargs)
  File "/home/pi/dingdang/client/notifier.py", line 41, in gather
    [client.run() for client in self.notifiers]
  File "/home/pi/dingdang/client/notifier.py", line 18, in run
    self.timestamp = self.gather(self.timestamp)
  File "/home/pi/dingdang/client/notifier.py", line 45, in handleEmailNotifications
    emails = Email.fetchUnreadEmails(self.profile, since=lastDate)
  File "/home/pi/dingdang/client/plugins/Email.py", line 141, in fetchUnreadEmails
    conn.login(profile[SLUG]['address'], profile[SLUG]['password'])
  File "/usr/lib/python2.7/imaplib.py", line 519, in login
    raise self.error(dat[-1])
error: LOGIN Login error

扫码后报错，无法进行声音识别，请看一下是什么问题

wzpan, 2017-08-05:
邮箱没有配置好，登录失败。请参考wiki的配置教程。

rekcah1986, 2017-08-05:
能用了，但是它一直给我播放方大同……

closed 2017-08-05 by rekcah1986

wzpan, 2017-08-05:
需要更新下contrib