wzpan/dingdang-robot #19: Create 微信登陆扫码网页（启动服务器之后的页面）

GoldJohnKing, 2017-08-15:
从下午两点开始折腾怎么git push，折腾到现在也没整明白，放弃了。
这玩意是放这里提还是放contrib提我也没想明白，但是看目录结构应该是这的。

下载地址： http://pan.baidu.com/s/1o8FRzzO 密码: smpt

预览图：
[image]

wzpan, 2017-08-15:
应该用 pull request 。
never mind，我回头看看加上。看起来挺漂亮的，赞一个。

wzpan, 2017-08-22:
已集成。需同时更新 dingdang 和 contrib 后才可用。页面真是很漂亮，给你的用心赞一个！

GoldJohnKing, 2017-08-22:
更新之后出现如下故障：

```
请扫描如下二维码登录微信
登录成功后，可以与自己的微信账号（不是文件传输助手）交互
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/dingdang/dingdang.py", line 132, in start_wxbot
    self.wxBot.run(self.mic)
  File "/home/pi/dingdang/client/wxbot.py", line 1177, in run
    self.gen_qr_code(os.path.join(self.login_pwd,'wxqr.png'))
  File "/home/pi/dingdang/client/wxbot.py", line 1233, in gen_qr_code
    qr.png(qr_file_path, scale=8)
  File "/usr/local/lib/python2.7/dist-packages/pyqrcode/__init__.py", line 463, in png
    module_color, background, quiet_zone)
  File "/usr/local/lib/python2.7/dist-packages/pyqrcode/builder.py", line 1383, in _png
    f, autoclose = _get_writable(file, 'wb')
  File "/usr/local/lib/python2.7/dist-packages/pyqrcode/builder.py", line 924, in _get_writable
    stream_or_path = open(stream_or_path, mode)
IOError: [Errno 2] No such file or directory: '/home/pi/login/wxqr.png'
```

wzpan, 2017-08-23:
主工程升级了没？

GoldJohnKing, 2017-08-23:
确定都升级了。貌似是wxbot无法下载二维码。

wzpan, 2017-08-23:
看了报错是试图生成二维码到 /home/pi/login 目录下。而我要求的是 /home/pi/dingdang/login 。
先试试cd dingdang然后再 python dingdang.py 看看有没有错误？

GoldJohnKing, 2017-08-23:
`python /home/pi/dingdang/dingdang.py`不可以，故障如前，但是
```
cd /home/pi/dingdang
python dingdang.py
```
可以，没有问题。

wzpan, 2017-08-23:
嗯，我找时间看看为什么变成相对路径了。

wzpan, 2017-08-23:
@GoldJohnKing 已修复。wxbot.py 获取路径的方式比较傻，我改了下。现在可以了。

closed 2017-08-24 by GoldJohnKing