wzpan/dingdang-robot #20: Wiki开机教程 创建软链接部分 存在问题

GoldJohnKing, 2017-08-18:
经测试，使用 `sudo ln -s /home/pi/.dingdang /root/.dingdang` 创建软链接，其实际位置位于 `/root/.dingdang/.dingdang` 。而要建立位于 /root 的软链接，应该使用 `sudo ln -s /home/pi/.dingdang /root/` （注意第二个root后面的`/`不能省略）。

请验证问题是否存在。
（我误删了`/root`之后重建软链接时发现的这个问题……）

wzpan, 2017-08-19:
亲测不存在此问题。我怀疑你把 `sudo ln -s /home/pi/.dingdang /root/.dingdang` 写成了 sudo ln -s /home/pi/.dingdang /root/.dingdang/` 了。
在没有创建 `/root/.dingdang` 这个目录前，是不可能创建一个目录到这个目录里头的。这是Linux里头的规定。
anyway，我改成你写的那句，避免你们打错。

GoldJohnKing, 2017-08-19:
为了以防万一是打错的问题……我还特地直接复制的教程里的命令来的……不过我的确证实这情况在我这确实发生了……上次改完教程我建议删除/root那次，回忆起来情况好像也和这个一样……

我的操作是(sudo略，此时有/root和软链接)：
rm -rf /root
(此时/root和软链接应该都已经没了……)
mkdir /root
ln -s /home/pi/.dingdang /root/.dingdang

然后就出现了上述情况。

anyway，没有问题是最好的……

closed 2017-08-19 by GoldJohnKing