wzpan/dingdang-robot #21: 运行python dingdang.py提示错误

icfancy, 2017-08-19:
S805 debian，1GB DDR
信息如下：
Traceback (most recent call last):
  File "dingdang.py", line 16, in <module>
    from client.conversation import Conversation
  File "/home/kevin/dingdang/client/conversation.py", line 6, in <module>
    from drivers.pixels import pixels
  File "/home/kevin/dingdang/client/drivers/pixels.py", line 124, in <module>
    pixels = Pixels()
  File "/home/kevin/dingdang/client/drivers/pixels.py", line 21, in __init__
    self.dev = apa102.APA102(num_led=self.PIXELS_N)
  File "/home/kevin/dingdang/client/drivers/apa102.py", line 90, in __init__
    self.spi.open(bus, device)  # Open SPI port 0, slave device (CS) 1
IOError: [Errno 2] No such file or directory

wzpan, 2017-08-19:
请参见[troubleshooting](https://github.com/wzpan/dingdang-robot/wiki/troubleshooting)

wzpan, 2017-08-22:
如 [troubleshooting](https://github.com/wzpan/dingdang-robot/wiki/troubleshooting) 里所述，在 raspi-config 里头开启 SPI 接口即可解决本问题。关闭本 issue 。

closed 2017-08-22 by wzpan