wzpan/dingdang-robot #25: 启动后报错：TypeError: argument of type 'NoneType' is not iterable

haofunny, 2017-08-23:
安装的镜像是8.13版本，
启动后出现以下问题：
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
INFO:client.stt:PocketSphinx 识别到了：[None]
Traceback (most recent call last):
  File "dingdang.py", line 190, in <module>
    app.run()
  File "dingdang.py", line 156, in run
    conversation.handleForever()
  File "/home/pi/dingdang/client/conversation.py", line 64, in handleForever
    threshold, transcribed = self.mic.passiveListen(self.persona)
  File "/home/pi/dingdang/client/mic.py", line 245, in passiveListen
    if any(PERSONA in phrase for phrase in transcribed):
  File "/home/pi/dingdang/client/mic.py", line 245, in <genexpr>
    if any(PERSONA in phrase for phrase in transcribed):
TypeError: argument of type 'NoneType' is not iterable

请教该如何解决，本人小白一枚

wzpan, 2017-08-23:
请更新下再试试

haofunny, 2017-08-24:
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
没接收到唤醒指令
没接收到唤醒指令
INFO:client.stt:PocketSphinx 识别到了：['']
INFO:client.stt:PocketSphinx 识别到了：['']
INFO:client.stt:PocketSphinx 识别到了：['']
INFO:client.stt:PocketSphinx 识别到了：['']
INFO:client.stt:PocketSphinx 识别到了：[None]
Traceback (most recent call last):
  File "dingdang.py", line 190, in <module>
    app.run()
  File "dingdang.py", line 156, in run
    conversation.handleForever()
  File "/home/pi/dingdang/client/conversation.py", line 64, in handleForever
    threshold, transcribed = self.mic.passiveListen(self.persona)
  File "/home/pi/dingdang/client/mic.py", line 245, in passiveListen
    if any(PERSONA in phrase for phrase in transcribed):
  File "/home/pi/dingdang/client/mic.py", line 245, in <genexpr>
    if any(PERSONA in phrase for phrase in transcribed):
TypeError: argument of type 'NoneType' is not iterable
还是显示这个问题

wzpan, 2017-08-27:
用的什么唤醒词？先试试用默认的。

wzpan, 2017-08-27:
加了个对 transcribed 为 None 情况下的例外。不过猜测这种情况是由于麦克风没有配置好导致。

haofunny, 2017-08-30:
好的，成功唤醒了，谢谢

closed 2017-08-30 by wzpan