wzpan/dingdang-robot #39: logo征集

wzpan, 2017-09-13:
征集叮当机器人的logo，将被用在项目主页、QQ群头像、论坛等多处。

欢迎贡献~~~

GoldJohnKing, 2017-09-14:
先报个名……最近事多……等稳定下来之后就尽力做一个……

顺路提一下……下次准备做镜像之前麻烦在群里说一声……我把给叮当搞的启动时自动更新的东西提个issue上来。

GoldJohnKing, 2017-09-17:
先贴一下目前的几个方案的草图。是草图，定夺之后还要按比例和尺寸重画一下。要不缩放之后的清晰度相当惨淡……

prefer方案：
[image]

4号方案：
[image]

0号方案：
[image]

1号方案：
[image]

2号方案：
[image]

3号方案：
脑洞太大，被我掐死了。

GoldJohnKing, 2017-09-17:
方案4带字样透明精修版：
[image]

wzpan, 2017-09-17:
ing和ang黑色吧，另外字弄小一点。再单独给一张没字的

closed 2017-09-19 by wzpan