wzpan/dingdang-robot #47: 关于叮当开机自启前增加自动升级的建议

GoldJohnKing, 2017-09-18:
如果希望叮当开机自启前自动升级，可以用以下文件内容替代已有的"dingdang.sh"文件中的内容。

对于希望以root身份启动叮当的用户：
`dingdang-AutoUpdate-Launcher-for-root-User.sh`

> #!/bin/bash
> sleep 1
> 
> #Delete Cache
> sudo rm -r /root/.cache
> sudo rm -r /root/.config
> sudo rm -r /root/.netease-musicbox
> sudo rm -r /root/userInfo
> sleep 1
> 
> #Update dingdang-robot
> cd /home/pi/dingdang
> git pull
> 
> #Update dingdang-contrib
> cd /home/pi/.dingdang/contrib
> git pull
> 
> #Update dingdang-contrib Requirements
> sudo pip install --upgrade -r requirements.txt
> sleep 1
> 
> #Restore Configuration of AlsaMixer
> alsactl --file=/home/pi/asound.state restore
> sleep 1
> 
> #Launch Dingdang in LxTerminal
> sudo lxterminal -e "python /home/pi/dingdang/dingdang.py"
> sleep 1
> 
> #Start Respeaker-Switcher in Background
> sudo python /home/pi/ReSpeaker-Switcher/switcher.py &

对于希望以pi身份启动叮当的用户：
`dingdang-AutoUpdate-Launcher-for-pi-User.sh`

> #!/bin/bash
> sleep 1
> 
> #Delete Cache
> sudo rm -r /home/pi/.cache
> sudo rm -r /home/pi/.config
> sudo rm -r /home/pi/.netease-musicbox
> sudo rm -r /home/pi/userInfo
> sleep 1
> 
> #AutoUpdate Before Launch
> #Update dingdang-robot
> cd /home/pi/dingdang
> git pull
> 
> #Update dingdang-contrib
> cd /home/pi/.dingdang/contrib
> git pull
> 
> #Update dingdang-contrib Requirements
> sudo pip install --upgrade -r requirements.txt
> sleep 1
> 
> #Restore Configuration of AlsaMixer
> alsactl --file=/home/pi/asound.state restore
> sleep 1
> 
> #Launch Dingdang in LxTerminal
> lxterminal -e "python /home/pi/dingdang/dingdang.py"
> sleep 1
> 
> #Start Respeaker-Switcher in Background
> sudo python /home/pi/ReSpeaker-Switcher/switcher.py &

另外，个人建议在叮当启动前清除其缓存，因为测试发现网易云音乐播放插件在某些情况下可能由于缓存问题导致默认播放列表长期不刷新。具体如下：
`dingdang-Launch-without-Update.sh`

> 
> #!/bin/bash
> sleep 1
> 
> #Delete Cache
> sudo rm -r /root/.cache
> sudo rm -r /root/.config
> sudo rm -r /root/.netease-musicbox
> sudo rm -r /root/userInfo
> 
> #Restore Configuration of AlsaMixer 
> alsactl --file=/home/pi/asound.state restore
> sleep 1
> 
> #Start DingDang
> sudo lxterminal -e "python /home/pi/dingdang/dingdang.py"
> sleep 1
> sudo python /home/pi/ReSpeaker-Switcher/switcher.py &

强烈建议将开机前自动升级功能加入到下一版本的镜像中，同时建议增加关于这一方面的Wiki内容。

wzpan, 2017-09-24:
wiki 已加上对本 issue 的引用。

新版本的叮当计划10月份发布。

wzpan, 2017-10-28:
这几个脚本已被集成至dingdang的 [launcher](https://github.com/wzpan/dingdang-robot/tree/master/launcher) 目录里头。我做了少量调整：

- 使用 tmux 代替 lxterminal 。好处是 ssh 进去系统的，可以通过 `tmux attach` 回到叮当的执行界面 ；
- 使用 `$HOME` 代替 `/home/pi` ，兼容其他主板用户；
- 执行如下两个命令前，都先检查下对应的文件/目录是否存在：

```
alsactl --file=/home/pi/asound.state restore

sudo python /home/pi/ReSpeaker-Switcher/switcher.py &
```
- 不删除 `$HOME/.config` ，因为启动脚本就放在里头……

closed 2017-11-04 by wzpan