wzpan/dingdang-robot #52: 按照手动安装指南安装失败

HughLi, 2017-09-28:
#### 由于我不擅长这个,现在按照wiki手动安装指南卡在了编译安装openFast这里,报错如下,特向各位大大请教下
> raspberry pi 3B
> OS: Raspbian 9.1 stretch
> Kernel: armv7l Linux 4.9.41-v7+

***
```sh
libtool: install: /usr/bin/install -c .libs/libfst.lai /usr/local/lib/libfst.la
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/local/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the `-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the `LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the `LD_RUN_PATH' environment variable
     during linking
   - use the `-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to `/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/home/pi/Downloads/openfst-1.3.4/src/lib'
make[2]: Leaving directory '/home/pi/Downloads/openfst-1.3.4/src/lib'
Making install in script
make[2]: Entering directory '/home/pi/Downloads/openfst-1.3.4/src/script'
/bin/bash ../../libtool --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H   -I./../include    -g -O2 -MT text-io.lo -MD -MP -MF .deps/text-io.Tpo -c -o text-io.lo text-io.cc
libtool: compile:  g++ -DHAVE_CONFIG_H -I./../include -g -O2 -MT text-io.lo -MD -MP -MF .deps/text-io.Tpo -c text-io.cc  -fPIC -DPIC -o .libs/text-io.o
text-io.cc: In function 'bool fst::script::WritePotentials(const string&, const std::vector<fst::script::WeightClass>&)':
text-io.cc:87:15: error: cannot convert 'std::ostream {aka std::basic_ostream<char>}' to 'bool' in initialization
   bool ret = *strm;
               ^~~~
Makefile:388: recipe for target 'text-io.lo' failed
make[2]: *** [text-io.lo] Error 1
make[2]: Leaving directory '/home/pi/Downloads/openfst-1.3.4/src/script'
Makefile:260: recipe for target 'install-recursive' failed
make[1]: *** [install-recursive] Error 1
make[1]: Leaving directory '/home/pi/Downloads/openfst-1.3.4/src'
Makefile:312: recipe for target 'install-recursive' failed
make: *** [install-recursive] Error 1
```

wzpan, 2017-09-28:
参照这个issue：https://github.com/kaldi-asr/kaldi/issues/941
应该是由于你的系统是 stretch，g++ 版本比较新不支持编译 1.3.4 版本的 openfst 。
试试换用 1.4.1 版本的 openfst 。

HughLi, 2017-09-28:
谢谢 已经编译通过了 可以关闭该问题了 @wzpan

closed 2017-09-28 by HughLi