wzpan/dingdang-robot #60: ImportError: cannot import name InsecureRequestWarning

bertsir, 2017-10-17:
运行的时候报这个问题
No handlers could be found for logger "cmuclmtk"
Traceback (most recent call last):
  File "dingdang.py", line 12, in <module>
    from client import stt
  File "/root/dingdang/client/stt.py", line 22, in <module>
    from requests.packages.urllib3.exceptions import InsecureRequestWarning
ImportError: cannot import name InsecureRequestWarning

wzpan, 2017-10-17:
requests 版本可能太老了。更新下再试试。

``` py
pip install --upgrade requests
```

bertsir, 2017-10-18:
root@raspberrypi:~/dingdang# pip install --upgrade requests
http://mirrors.aliyun.com/pypi/simple/requests/ uses an insecure transport scheme (http). Consider using https if mirrors.aliyun.com has it available
Downloading/unpacking requests from http://mirrors.aliyun.com/pypi/packages/49/df/50aa1999ab9bde74656c2919d9c0c085fd2b3775fd3eca826012bef76d8c/requests-2.18.4-py2.py3-none-any.whl#md5=eb9be71cc41fd73a51a7c9cd1adde5de
  Downloading requests-2.18.4-py2.py3-none-any.whl (88kB): 88kB downloaded
http://mirrors.aliyun.com/pypi/simple/certifi/ uses an insecure transport scheme (http). Consider using https if mirrors.aliyun.com has it available
Requirement already up-to-date: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests)
http://mirrors.aliyun.com/pypi/simple/idna/ uses an insecure transport scheme (http). Consider using https if mirrors.aliyun.com has it available
Requirement already up-to-date: idna>=2.5,<2.7 in /usr/local/lib/python2.7/dist-packages (from requests)
Downloading/unpacking chardet>=3.0.2,<3.1.0 (from requests)
  http://mirrors.aliyun.com/pypi/simple/chardet/ uses an insecure transport scheme (http). Consider using https if mirrors.aliyun.com has it available
  Downloading chardet-3.0.4-py2.py3-none-any.whl (133kB): 133kB downloaded
Downloading/unpacking urllib3>=1.21.1,<1.23 (from requests)
  http://mirrors.aliyun.com/pypi/simple/urllib3/ uses an insecure transport scheme (http). Consider using https if mirrors.aliyun.com has it available
  Downloading urllib3-1.22-py2.py3-none-any.whl (132kB): 132kB downloaded
Installing collected packages: requests, chardet, urllib3
  Found existing installation: requests 2.4.3
    Not uninstalling requests at /usr/lib/python2.7/dist-packages, owned by OS
  Found existing installation: chardet 2.3.0
    Not uninstalling chardet at /usr/lib/python2.7/dist-packages, owned by OS
  Found existing installation: urllib3 1.9.1
    Not uninstalling urllib3 at /usr/lib/python2.7/dist-packages, owned by OS
Successfully installed requests chardet urllib3
Cleaning up...
root@raspberrypi:~/dingdang# python dingdang.py 
No handlers could be found for logger "cmuclmtk"
Traceback (most recent call last):
  File "dingdang.py", line 12, in <module>
    from client import stt
  File "/root/dingdang/client/stt.py", line 22, in <module>
    from requests.packages.urllib3.exceptions import InsecureRequestWarning
ImportError: cannot import name InsecureRequestWarning

更新之后还是不行

bertsir, 2017-10-18:
问题解决了 是我pip是用apt-get安装的  树莓派上apt-get安装的pip无法升级request  卸载掉了pip换脚本安装方式重新安装pip  问题解决

closed 2017-10-18 by bertsir