wzpan/dingdang-robot #68: ImportError: No module named pyaudio

Osub, 2017-11-15:
启动安装失败
ImportError: No module named pyaudio

wzpan, 2017-11-15:
请认真阅读 [手动安装](https://github.com/wzpan/dingdang-robot/wiki/install#%E6%89%8B%E5%8A%A8%E5%AE%89%E8%A3%85) 一节。最开头已经要求安装 `python-pyaudio` 等依赖了。

Osub, 2017-11-15:
我用的是 centos    @wzpan

Osub, 2017-11-15:
```
python -m pip install pyaudio
Collecting pyaudio
  Using cached PyAudio-0.2.11.tar.gz
Building wheels for collected packages: pyaudio
  Running setup.py bdist_wheel for pyaudio ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Qw6gNx/pyaudio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp4BA3eipip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-2.7
  copying src/pyaudio.py -> build/lib.linux-armv7l-2.7
  running build_ext
  building '_portaudio' extension
  creating build/temp.linux-armv7l-2.7
  creating build/temp.linux-armv7l-2.7/src
  gcc -pthread -fno-strict-aliasing -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=hard -D_GNU_SOURCE -fPIC -fwrapv -DNDEBUG -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=hard -D_GNU_SOURCE -fPIC -fwrapv -fPIC -I/usr/include/python2.7 -c src/_portaudiomodule.c -o build/temp.linux-armv7l-2.7/src/_portaudiomodule.o
  In file included from /usr/include/python2.7/pyconfig.h:4:0,
                   from /usr/include/python2.7/Python.h:8,
                   from src/_portaudiomodule.c:28:
  /usr/include/python2.7/pyconfig-32.h:1188:0: 警告：“_POSIX_C_SOURCE”重定义 [默认启用]
   #define _POSIX_C_SOURCE 200112L
   ^
  In file included from /usr/include/stdio.h:27:0,
                   from src/_portaudiomodule.c:27:
  /usr/include/features.h:168:0: 附注：这是先前定义的位置
   # define _POSIX_C_SOURCE 200809L
   ^
  In file included from /usr/include/python2.7/pyconfig.h:4:0,
                   from /usr/include/python2.7/Python.h:8,
                   from src/_portaudiomodule.c:28:
  /usr/include/python2.7/pyconfig-32.h:1210:0: 警告：“_XOPEN_SOURCE”重定义 [默认启用]
   #define _XOPEN_SOURCE 600
   ^
  In file included from /usr/include/stdio.h:27:0,
                   from src/_portaudiomodule.c:27:
  /usr/include/features.h:170:0: 附注：这是先前定义的位置
   # define _XOPEN_SOURCE 700
   ^
  src/_portaudiomodule.c:29:23: 致命错误：portaudio.h：没有那个文件或目录
   #include "portaudio.h"
                         ^
  编译中断。
  error: command 'gcc' failed with exit status 1

  ----------------------------------------
  Failed building wheel for pyaudio
  Running setup.py clean for pyaudio
Failed to build pyaudio
Installing collected packages: pyaudio
  Running setup.py install for pyaudio ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Qw6gNx/pyaudio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-9Smp2h-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    copying src/pyaudio.py -> build/lib.linux-armv7l-2.7
    running build_ext
    building '_portaudio' extension
    creating build/temp.linux-armv7l-2.7
    creating build/temp.linux-armv7l-2.7/src
    gcc -pthread -fno-strict-aliasing -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=hard -D_GNU_SOURCE -fPIC -fwrapv -DNDEBUG -O2 -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=hard -D_GNU_SOURCE -fPIC -fwrapv -fPIC -I/usr/include/python2.7 -c src/_portaudiomodule.c -o build/temp.linux-armv7l-2.7/src/_portaudiomodule.o
    In file included from /usr/include/python2.7/pyconfig.h:4:0,
                     from /usr/include/python2.7/Python.h:8,
                     from src/_portaudiomodule.c:28:
    /usr/include/python2.7/pyconfig-32.h:1188:0: 警告：“_POSIX_C_SOURCE”重定义 [默认启用]
     #define _POSIX_C_SOURCE 200112L
     ^
    In file included from /usr/include/stdio.h:27:0,
                     from src/_portaudiomodule.c:27:
    /usr/include/features.h:168:0: 附注：这是先前定义的位置
     # define _POSIX_C_SOURCE 200809L
     ^
    In file included from /usr/include/python2.7/pyconfig.h:4:0,
                     from /usr/include/python2.7/Python.h:8,
                     from src/_portaudiomodule.c:28:
    /usr/include/python2.7/pyconfig-32.h:1210:0: 警告：“_XOPEN_SOURCE”重定义 [默认启用]
     #define _XOPEN_SOURCE 600
     ^
    In file included from /usr/include/stdio.h:27:0,
                     from src/_portaudiomodule.c:27:
    /usr/include/features.h:170:0: 附注：这是先前定义的位置
     # define _XOPEN_SOURCE 700
     ^
    src/_portaudiomodule.c:29:23: 致命错误：portaudio.h：没有那个文件或目录
     #include "portaudio.h"
                           ^
    编译中断。
    error: command 'gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Qw6gNx/pyaudio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-9Smp2h-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Qw6gNx/pyaudio/
```

wzpan, 2017-11-15:
使用其他操作系统的不在我的支持范围内，请自行解决依赖安装问题。

closed 2017-11-17 by wzpan