wzpan/dingdang-robot #71: 百度tts和stt的token可以本地缓存一下

xizirumeng, 2017-11-22:
```
import time
from dateutil import parser

def get_token(self):
    cache = open(dingdangpath.TEMP_PATH+'/baidustt.ini' , 'a+')
            try:
                pms = cache.readlines()
                if len(pms) > 0:
                    time = pms[0]
                    tk = pms[1]
                    # 计算token是否过期 86400
                    time = parser.parse(time)
                    endtime = datetime.datetime.now()
                    if (endtime - time).seconds <= 86300:
                        return tk
            finally:
                cache.close()
    URL = 'http://openapi.baidu.com/oauth/2.0/token'
            params = urllib.urlencode({'grant_type': 'client_credentials',
                                       'client_id': self.api_key,
                                       'client_secret': self.secret_key})
            r = requests.get(URL, params=params)
            try:
                r.raise_for_status()
                token = r.json()['access_token']
                # 存储token
                try:
                    cache = open(dingdangpath.TEMP_PATH+'/baidustt.ini' , 'w')
                    cache.write(str(datetime.datetime.now()) + '\n')
                    cache.write(token)
                finally:
                    cache.close()
                return token
            except requests.exceptions.HTTPError:
                self._logger.critical('Token request failed with response: %r',
                                      r.text,
                                      exc_info=True)
                return ''

```

wzpan, 2017-11-23:
不错，回头试试。

wzpan, 2017-11-26:
done

closed 2017-11-28 by wzpan