wzpan/dingdang-robot #73: 网易云音乐找不到正确的音乐url

jifengli, 2017-11-23:
请问网易云音乐插件可以正常使用么？我的每次都会提示音乐文件不存在，我复制音乐url到浏览器也显示找不到文件，说明插件里获得的url是不正确的．请问如何解决？是不是网易云音乐API又改了？

shark526, 2017-11-23:
遇到同样问题，不知道默认登陆进去播放的是哪个歌单的音乐，我在想是不是因为那首歌的链接过期了， 如果是这样找到默认歌单就好办了

wzpan, 2017-11-23:
网易云音乐api已经变了，暂时播不了

shark526, 2017-11-24:
@wzpan 叮当用的和这个相关吗？https://github.com/darknessomi/musicbox， 这个还可以用。

zanjie1999, 2017-11-24:
可以试试用musicbox播放音乐…感觉还蛮方便，可以和叮当一起运行不会冲突

zanjie1999, 2017-11-26:
更新： musicbox也炸了

SkyAzure, 2017-12-22:
`musicbox`2017-11-28 版本 0.2.4.2 更新获取歌曲列表的 api

wzpan, 2017-12-24:
@SkyAzure 测试了一下musicbox，拿到的地址还是错的……你确定能播？

zanjie1999, 2017-12-27:
之前那个接口好像恢复正常了,我写的那个脚本已经可以正常的拿到没歌单了

lzj1530069991, 2018-01-06:
如何将musicbox用语音控制呢，或者说怎么把musicbox移植到dingdang里呢

zanjie1999, 2018-01-07:
今天发现musicbox好像播不了音乐了?各种抛异常

wzpan, 2018-01-07:
@zanjie1999 没错，网易云音乐也挂了。我正在封一个新的接口，如果可以会提供一个新的插件出来。

shark526, 2018-01-08:
@zanjie1999 昨晚还试了， 没毛病啊。

zanjie1999, 2018-01-08:
'http://music.163.com/song/media/outer/url?id=' + tracksId + '.mp3' @wzpan

wzpan, 2018-01-08:
好，我晚点试试 @zanjie1999

wzpan, 2018-01-08:
已修复

closed 2018-01-14 by wzpan