wzpan/dingdang-robot #78: 執行後報錯

jl000046, 2017-12-01:
raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='proxy.peiyinge.com', port=17063): Max retries exceeded with url: /synth?ts=1512114723&sign=30859d3464a3ae1a155598ac4d255c0c&vid=67100&volume=&speed=0&content=%E4%BC%9F%E6%B4%B2%20%E6%88%91%E8%83%BD%E4%B8%BA%E6%82%A8%E5%81%9A%E4%BB%80%E4%B9%88%3F (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x741f1890>: Failed to establish a new connection: [Errno 111] Connection refused',))

用的是自己的訊飛科大ID及API KEY，
但仍出現上列錯誤，連線遭拒，
求解，謝謝！！

wzpan, 2017-12-02:
@jl000046 是否还有这个问题？

jl000046, 2017-12-03:
@wzpan 是，仍然有此問題。

wzpan, 2017-12-03:
@jl000046 是否修改过 vid 的取值？如果是，改成其他的试试。如果不是，试试换成其他的TTS服务。

jl000046, 2017-12-15:
換成百度語音即可正常運行，海外申請遇到了些麻煩，總之可以使用了，十分感謝！

closed 2017-12-15 by jl000046