wzpan/dingdang-robot #83: Ubuntu 16.04.3 LTS 下安装不通过

caohm, 2017-12-12:
Ubuntu 16.04.3 LTS

sudo apt-get install libportaudio-dev 不过
编译安装 Phonetisaurus 不过

wzpan, 2017-12-13:
编译不过请先贴具体原因。
另外，Linux 发行版本太多，每个版本的包千差万别，所以 dingdang-robot 官方只保证能在 Raspbian 系统上能顺利安装运行。其他系统主要靠用户自行解决编译安装问题。

caohm, 2017-12-13:
libportaudio 安装的 libportaudio19
is2013-conversion/phonetisaurus/src/Makefile line:43 “EXTRA=$(CFLAGS) $(THIRD_PARTIES_INCLUDE) -std=c++11 -O3 $(CPPFLAGS)”编译通过
client/conversation.py pixels去掉
client/mic.py pixels去掉
python dingdang.py 起动成功

wzpan, 2017-12-14:
> client/conversation.py pixels去掉
> client/mic.py pixels去掉

master 分支并没有依赖 pixels ，是不是拉成 respeaker 分支了？

caohm, 2017-12-14:
嗯是的

closed 2017-12-16 by wzpan

mindcont, 2018-02-02:
正如@caohm 说得那样，通过在`is2013-conversion/phonetisaurus/src/Makefile` Line 42 中 指定使用`c++11`编译标准，可以编译通过。 
```java
THIRD_PARTIES_INCLUDE ?= -I3rdparty/sparsehash -I3rdparty/utfcpp
# EXTRA=$(CFLAGS) $(THIRD_PARTIES_INCLUDE) $(CXXFLAGS) $(CPPFLAGS)
EXTRA=$(CFLAGS) $(THIRD_PARTIES_INCLUDE) -std=c++11 -O3 $(CPPFLAGS)

POBJS=Phonetisaurus.o FstPathFinder.o MBRDecoder.o LatticePruner.o util.o
MOBJS=M2MFstAligner.o FstPathFinder.o MBRDecoder.o LatticePruner.o util.o
OOBJS=PhonetisaurusOmega.o PhonetisaurusE2F.o util.o FstPathFinder.o LatticePruner.o MBRDecoder.o
```

更多的也可以参考最新的[Phonetisaurus](https://github.com/AdolfVonKleist/Phonetisaurus).