wzpan/dingdang-robot #85: 镜像安装 ImportError: cannot import name get_localzone

marklin0531, 2017-12-16:
您好,我是用鏡像安裝,但在啟動叮噹時,發生下面的錯誤,
請問要如何解決呢??
謝謝.

pi@raspberrypi:~/dingdang $ python dingdang.py
Traceback (most recent call last):
  File "dingdang.py", line 16, in <module>
    from client.conversation import Conversation
  File "/home/pi/dingdang/client/conversation.py", line 3, in <module>
    from notifier import Notifier
  File "/home/pi/dingdang/client/notifier.py", line 5, in <module>
    from apscheduler.schedulers.background import BackgroundScheduler
  File "/usr/local/lib/python2.7/dist-packages/apscheduler/schedulers/background.py", line 4, in <module>
    from apscheduler.schedulers.base import BaseScheduler
  File "/usr/local/lib/python2.7/dist-packages/apscheduler/schedulers/base.py", line 10, in <module>
    from tzlocal import get_localzone
ImportError: cannot import name get_localzone

wzpan, 2017-12-16:
未能复现。试试手动更新下 apscheduler：

``` sh
sudo pip install --upgrade apscheduler
```

marklin0531, 2017-12-17:
重裝後還是出錯, 
我手動安裝系統看看. 
謝謝

wzpan, 2017-12-24:
@marklin0531 解决了吗？

marklin0531, 2017-12-24:
後來重新手動安裝系統
沒用鏡像的方式了
算解決了 謝謝你

closed 2017-12-24 by wzpan