wzpan/dingdang-robot #86: 微信语音消息无反应

SkyAzure, 2017-12-21:
非常感谢您及众多贡献者的付出，才让我能实现自己制作智能音箱

1. 登录微信后，可以对文本消息进行回复，`[echo]`功能也正常，但是发送语音没有反应
2. 如何在宿舍中实现舍友也可以使用微信的`[echo]`功能，当其他特定好友给本人发送消息时也激活回传功能

wzpan, 2017-12-22:
1. 微信的语音是用来控制叮当还是当做echo功能，是由配置决定的。

```
# 当微信发送语音时，是直接播放语音还是执行语音命令？
# true：直接播放
# false：执行语音命令（只支持百度STT，其他两种STT识别不准）
wechat_echo: false
```

2. 想法不错，也可以做到。我可以实现一下。

SkyAzure, 2017-12-22:
1. 配置是wechat_echo: true  发送语音消息，报错为
   `TypeError: Can't instantiate abstract class SimpleMp3Player with abstract methods get speech`
2. 希望大佬帮助实现这个功能

wzpan, 2017-12-24:
1. 问题已修复，是 #84 这个 pull-request 引入的 bug；
2. 已实现。profile.yml 新增两个配置项，支持控制 echo 的白名单：

``` yml
# 除了自己之外，还能响应 echo 指令的好友微信名单
wechat_echo_text_friends: ['小Q', 'HaHack']
# 除了自己之外，还能直接播放语音的好友微信名单
wechat_echo_voice_friends: ['小Q']
```

SkyAzure, 2017-12-25:
感谢大佬

closed 2017-12-25 by wzpan