wzpan/dingdang-robot #88: 进程运行中退出

yasing, 2017-12-24:
Hi,
    经过手动安装，基本能够使用，但是有几次就出现突然退出的情况。LOG如下。这类问题大致是什么原因引起的呢？
2017-12-24 19:03:58,425 mic.py[line:336]         INFO 机器人说：已取消
2017-12-24 19:03:58,435 connectionpool.py[line:188]         INFO Starting new HTTP connection (1): tsn.baidu.com
2017-12-24 19:04:09,055 base.py[line:106]         INFO Running job "Notifier.gather (trigger: interval[0:00:30], next run at: 2017-12-24 19:04:09 HKT)" (scheduled at 2017-12-24 19:04:09.046493+08:00)
2017-12-24 19:04:09,056 base.py[line:117]         INFO Job "Notifier.gather (trigger: interval[0:00:30], next run at: 2017-12-24 19:04:09 HKT)" executed successfully
2017-12-24 19:04:12,075 brain.py[line:113]         ERROR Failed to execute plugin
Traceback (most recent call last):
  File "/home/pi/dingdang/client/brain.py", line 109, in query
    plugin.handle(text, self.mic, self.profile, wxbot)
  File "/home/pi/.dingdang/contrib/Direction.py", line 28, in handle
    mic.say(u'已取消')
  File "/home/pi/dingdang/client/mic.py", line 345, in say
    self.speaker.say(phrase, cache)
  File "/home/pi/dingdang/client/tts.py", line 123, in say
  File "/home/pi/dingdang/client/tts.py", line 552, in get_speech
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 99, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 49, in request
    response = session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 461, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 573, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/adapters.py", line 415, in send
    raise ConnectionError(err, request=request)
ConnectionError: ('Connection aborted.', BadStatusLine("''",))
2017-12-24 19:04:12,091 mic.py[line:336]         INFO 机器人说：抱歉，我的大脑出故障了，晚点再试试吧
2017-12-24 19:04:12,101 connectionpool.py[line:188]         INFO Starting new HTTP connection (1): tsn.baidu.com
2017-12-24 19:04:23,078 stt.py[line:190]         INFO PocketSphinx 识别到了：['PAIPAI']
2017-12-24 19:04:23,080 conversation.py[line:71]         INFO Keyword 'PAIPAI' has been said!
2017-12-24 19:04:27,644 connectionpool.py[line:188]         INFO Starting new HTTP connection (1): vop.baidu.com
2017-12-24 19:04:39,055 base.py[line:106]         INFO Running job "Notifier.gather (trigger: interval[0:00:30], next run at: 2017-12-24 19:04:39 HKT)" (scheduled at 2017-12-24 19:04:39.046493+08:00)
2017-12-24 19:04:39,056 base.py[line:117]         INFO Job "Notifier.gather (trigger: interval[0:00:30], next run at: 2017-12-24 19:04:39 HKT)" executed successfully
2017-12-24 19:04:40,545 base.py[line:159]         INFO Scheduler has been shut down

wzpan, 2017-12-24:
看样子是在调用百度的 TTS  时出了问题。试试换成其他 TTS 服务？

yasing, 2017-12-25:
换成讯飞的暂时没有出现这种问题。感谢！

closed 2017-12-25 by yasing