wzpan/dingdang-robot #95: 运行报错，ImportError: libpocketsphinx.so.1: cannot open shared object file

Cdmoonb, 2018-01-04:
2018-01-04 19:01:54,033 dingdang.py[line:174]             ERROR Error occured!
Traceback (most recent call last):
  File "dingdang.py", line 172, in <module>
    app = Dingdang()
  File "dingdang.py", line 94, in __init__
    stt_passive_engine_class.get_passive_instance(),
  File "/home/chenpeng/dingdang/client/stt.py", line 59, in get_passive_instance
    return cls.get_instance('keyword', phrases)
  File "/home/chenpeng/dingdang/client/stt.py", line 53, in get_instance
    instance = cls(**config)
  File "/home/chenpeng/dingdang/client/stt.py", line 110, in __init__
    import pocketsphinx as ps
ImportError: libpocketsphinx.so.1: cannot open shared object file: No such file or directory

wzpan, 2018-01-14:
没装好 pocketsphinx ，试过安装对应的 python 库没？

sudo apt-get install python-pockersphinx

Jaylon, 2018-01-15:
pi@raspberrypi:~/workspace/dingdang $ cat temp/dingdang.log
2018-01-15 23:53:32,822 dingdang.py[line:175]             ERROR Error occured!
Traceback (most recent call last):
  File "dingdang.py", line 173, in <module>
    app = Dingdang()
  File "dingdang.py", line 95, in __init__
    stt_passive_engine_class.get_passive_instance(),
  File "/home/pi/workspace/dingdang/client/stt.py", line 66, in get_passive_instance
    return cls.get_instance('keyword', phrases)
  File "/home/pi/workspace/dingdang/client/stt.py", line 60, in get_instance
    instance = cls(**config)
  File "/home/pi/workspace/dingdang/client/stt.py", line 113, in __init__
    import pocketsphinx as ps
ImportError: libpocketsphinx.so.1: cannot open shared object file: No such file or directory

pi@raspberrypi:~/workspace/dingdang $ uname -a
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

---------------------------
遇到同样的问题。执行以下两条命令后，问题依旧没有解决。
 $ pip install --upgrade pip setuptools wheel
$ pip install --upgrade pocketsphinx

wzpan, 2018-01-16:
建议使用镜像安装方式，不同的系统差别太大，难以帮助解决。

Jaylon, 2018-01-16:
运行 sudo ldconfig 命令之后，问题解决。

Cdmoonb, 2018-01-17:
我重装了系统ubuntu16.04LTS 
然后重新按照步骤走了一遍
是成功的 
应该是我中间步骤出错了
谢

closed 2018-01-17 by Cdmoonb