wzpan/dingdang-robot #99: 启动后，报错错找不到文件

zjy45, 2018-01-11:
安装方式：手动安装
操作系统：Raspbian
问题描述：areocrd和aplay 正常，但启动后报错找不到文件，望指点，谢谢
复现步骤： 
```
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Traceback (most recent call last):
  File "dingdang/dingdang.py", line 177, in <module>
    app.run()
  File "dingdang/dingdang.py", line 123, in run
    self.mic.say(salutation, cache=True)
  File "/home/pi/dingdang/client/mic.py", line 347, in say
    self.speaker.say(phrase, cache)
  File "/home/pi/dingdang/client/tts.py", line 127, in say
    self.play_mp3(tmpfile)
  File "/home/pi/dingdang/client/tts.py", line 98, in play_mp3
    p = subprocess.Popen(cmd, stdout=f, stderr=f)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```

closed 2018-01-13 by zjy45

wzpan, 2018-01-14:
@zjy45 解决了吗？

zjy45, 2018-01-14:
@wzpan 之前论坛发帖说暂时封禁，然后我就在GitHub也问了问题，然后您回答了我论坛的问题，所以我就把GitHub的问题关了。
最新状况是：更新后还是有问题

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Traceback (most recent call last):
File “dingdang.py”, line 178, in 
app.run()
File “dingdang.py”, line 124, in run
self.mic.say(salutation, cache=True)
File “/home/pi/dingdang/client/mic.py”, line 348, in say
self.speaker.say(phrase, cache)
File “/home/pi/dingdang/client/tts.py”, line 135, in say
self.play_mp3(tmpfile)
File “/home/pi/dingdang/client/tts.py”, line 106, in play_mp3
p = subprocess.Popen(cmd, stdout=f, stderr=f)
File “/usr/lib/python2.7/subprocess.py”, line 390, in init
errread, errwrite)
File “/usr/lib/python2.7/subprocess.py”, line 1024, in _execute_child
raise child_exception
OSError: [Errno 2] No such file or directory

论坛问题地址：http://bbs.hahack.com/t/topic/163

wzpan, 2018-01-14:
换个tts服务再试试呢？可能是没有合成语音