wzpan/dingdang-robot #103: 手动安装，启动报错 OSError: [Errno 2] No such file or directory

Jaylon, 2018-01-16:
pi@raspberrypi:~/workspace/dingdang $ python dingdang.py

*******************************************************"
*             叮当 - 中文语音对话机器人               *
*          (c) 2017 潘伟洲 <m@hahack.com>             *
*   https://github.com/wzpan/dingdang-robot.git       *
*******************************************************

如需查看log，可以执行 `tail -f 叮当所在目录/temp/dingdang.log`

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Traceback (most recent call last):
  File "dingdang.py", line 178, in <module>
    app.run()
  File "dingdang.py", line 124, in run
    self.mic.say(salutation, cache=True)
  File "/home/pi/workspace/dingdang/client/mic.py", line 348, in say
    self.speaker.say(phrase, cache)
  File "/home/pi/workspace/dingdang/client/tts.py", line 135, in say
    self.play_mp3(tmpfile)
  File "/home/pi/workspace/dingdang/client/tts.py", line 106, in play_mp3
    p = subprocess.Popen(cmd, stdout=f, stderr=f)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

------
pi@raspberrypi:~/workspace/dingdang $ cat temp/dingdang.log
2018-01-16 21:39:12,885 mic.py[line:47]             INFO Initializing PyAudio. ALSA/Jack error messages that pop up during this process are normal and can usually be safely ignored.
2018-01-16 21:39:12,973 mic.py[line:54]             INFO Initialization of PyAudio completed.
2018-01-16 21:39:13,100 base.py[line:132]             INFO Scheduler started
2018-01-16 21:39:13,177 base.py[line:743]             INFO Added job "Notifier.gather" to job store "default"
2018-01-16 21:39:13,178 mic.py[line:339]             INFO 机器人说：伟洲 我能为您做什么?
2018-01-16 21:39:13,822 base.py[line:159]             INFO Scheduler has been shut down

-----
(在log 中有“伟洲 我能为您做什么?“，但是没有这个声音出现。)

wzpan, 2018-01-17:
用的是什么TTS服务？换一个试试？

Jaylon, 2018-01-18:
问题原因是sox没有安装，安装sox之后问题解决。

closed 2018-01-18 by Jaylon