wzpan/dingdang-robot #107: Makefile:51: recipe for target 'Phonetisaurus.o' failed

YountMan, 2018-01-19:
大佬好, 按照

wzpan, 2018-01-20:
什么版本的系统？

YountMan, 2018-01-21:
Duplicate of #103 pi@pi_ssd:~ $ uname -a
Linux pi_ssd 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
pi@pi_ssd:~ $ cat /proc/version
Linux version 4.9.35-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1014 SMP Fri Jun 30 14:47:43 BST 2017
pi@pi_ssd:~ $ cat /etc/issue
Raspbian GNU/Linux 8 \n \l

pi@pi_ssd:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
pi@pi_ssd:~ $

YountMan, 2018-01-21:
build 180121
解决了， Google 到的 <br/>
贴出地址 ： https://github.com/jasperproject/jasper-client/issues/267<br/>
变更为   <code>sudo  CPPFLAGS="-std=c++11" make</code> <br/>
make 通过
 还是 多谢@wzpan 大佬的回复

closed 2018-01-21 by wzpan