wzpan/dingdang-robot #112: 微信二维码无法登陆

kkycj, 2018-01-27:
前段时间有使用微信扫一扫相机扫描邮箱中的二维码图像无法登陆微信的情况，今天从宿舍搬回家，重新设置网络环境后，一使用语音发送二维码的命令叮当就没有响应，而且无法再次用语音唤醒？请问是什么问题？

kkycj, 2018-01-27:
好像不只是微信登陆功能 唤醒后接受一句语音命令就进入无法响应状态。。。

kkycj, 2018-01-27:
运行叮当后显示如下
`Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unloc                                                                                                             k
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unloc                                                                                                             k
`

wzpan, 2018-01-29:
不响应说明你的麦不行。

显示的log没有问题。

kkycj, 2018-01-31:
重启之后第一次用语音指令是可以唤醒的，但是在蜂鸣声之后没有响应，再往后也不能唤醒，请问也是麦的问题吗？用的是ps3 eye的设备。

wzpan, 2018-01-31:
关了微信能正常工作吗

kkycj, 2018-02-03:
不可以，只能使用微信发送命令，语音完全不能工作

wzpan, 2018-02-03:
语音不能工作说明没配好mic，看下 [常见问题解答](https://github.com/wzpan/dingdang-robot/wiki/troubleshooting#%E4%BD%BF%E7%94%A8%E9%97%AE%E9%A2%98)

olive900, 2018-02-10:
今天试了一下，扫微信二维码不能登陆。扫官网的可以登陆。

wzpan, 2018-02-10:
终端颜色主题请使用黑色底白色字，否则会扫不出来。另外有些地区可能登不上，这是微信网页客户端的api问题

wzpan, 2018-02-24:
楼上已经给出了建议，如果有问题可以进一步讨论。

closed 2018-02-24 by wzpan