wzpan/dingdang-robot #115: Phonetisaurus 编译失败

andy2080, 2018-02-02:
您好，我在nanopi上安装叮当编译Phonetisaurus的时候失败：

`(std::vector<std::pair<typename A::Label, typename A::Label> >*, bool)’:
/usr/local/include/fst/label-reachable.h:249:5: error: ‘unordered_map’ was not declared in this scope
     unordered_map<Label, Label> &label2index = *data_->Label2Index();
     ^
/usr/local/include/fst/label-reachable.h:249:24: error: expected primary-expression before ‘,’ token
     unordered_map<Label, Label> &label2index = *data_->Label2Index();
                        ^
/usr/local/include/fst/label-reachable.h:249:31: error: expected primary-expression before ‘>’ token
     unordered_map<Label, Label> &label2index = *data_->Label2Index();
                               ^
/usr/local/include/fst/label-reachable.h:249:34: error: ‘label2index’ was not declared in this scope
     unordered_map<Label, Label> &label2index = *data_->Label2Index();
                                  ^
/usr/local/include/fst/label-reachable.h:251:32: error: invalid type in declaration before ‘<’ token
     for (typename unordered_map<Label, Label>::const_iterator
                                ^
/usr/local/include/fst/label-reachable.h:251:32: error: expected ‘;’ before ‘<’ token
/usr/local/include/fst/label-reachable.h:251:32: error: expected primary-expression before ‘<’ token
/usr/local/include/fst/label-reachable.h:251:38: error: expected primary-expression before ‘,’ token
     for (typename unordered_map<Label, Label>::const_iterator
                                      ^
/usr/local/include/fst/label-reachable.h:251:45: error: expected primary-expression before ‘>’ token
     for (typename unordered_map<Label, Label>::const_iterator
                                             ^
/usr/local/include/fst/label-reachable.h:251:46: error: ‘::const_iterator’ has not been declared
     for (typename unordered_map<Label, Label>::const_iterator
                                              ^
/usr/local/include/fst/label-reachable.h:252:14: error: expected ‘;’ before ‘it’
              it = label2index.begin(); it != label2index.end(); ++it)
              ^
/usr/local/include/fst/label-reachable.h:252:14: error: ‘it’ was not declared in this scope
/usr/local/include/fst/label-reachable.h:252:38: error: expected ‘)’ before ‘;’ token
              it = label2index.begin(); it != label2index.end(); ++it)
                                      ^
/usr/local/include/fst/label-reachable.h:252:40: error: ‘it’ was not declared in this scope
              it = label2index.begin(); it != label2index.end(); ++it)
                                        ^
/usr/local/include/fst/label-reachable.h:259:18: error: expected nested-name-specifier before ‘unordered_map’
         typename unordered_map<Label, Label>::const_iterator
                  ^
/usr/local/include/fst/label-reachable.h:259:18: error: expected ‘(’ before ‘unordered_map’
/usr/local/include/fst/label-reachable.h: In member function ‘void fst::LabelReachable<A, S, D>::TransformFst()’:
/usr/local/include/fst/label-reachable.h:435:15: error: ‘label2state_’ was not declared in this scope
           if (label2state_.find(label) == label2state_.end()) {
               ^
/usr/local/include/fst/label-reachable.h:440:27: error: ‘label2state_’ was not declared in this scope
           arc.nextstate = label2state_[label];
                           ^
/usr/local/include/fst/label-reachable.h:449:13: error: ‘label2state_’ was not declared in this scope
         if (label2state_.find(kNoLabel) == label2state_.end()) {
             ^
/usr/local/include/fst/label-reachable.h:454:44: error: ‘label2state_’ was not declared in this scope
         Arc arc(kNoLabel, kNoLabel, final, label2state_[kNoLabel]);
                                            ^
/usr/local/include/fst/label-reachable.h: In member function ‘void fst::LabelReachable<A, S, D>::FindIntervals(fst::LabelReachable<A, S, D>::StateId)’:
/usr/local/include/fst/label-reachable.h:491:5: error: ‘unordered_map’ was not declared in this scope
     unordered_map<Label, Label> &label2index = *data_->Label2Index();
     ^
/usr/local/include/fst/label-reachable.h:491:24: error: expected primary-expression before ‘,’ token
     unordered_map<Label, Label> &label2index = *data_->Label2Index();
                        ^
/usr/local/include/fst/label-reachable.h:491:31: error: expected primary-expression before ‘>’ token
     unordered_map<Label, Label> &label2index = *data_->Label2Index();
                               ^
/usr/local/include/fst/label-reachable.h:491:34: error: ‘label2index’ was not declared in this scope
     unordered_map<Label, Label> &label2index = *data_->Label2Index();
                                  ^
/usr/local/include/fst/label-reachable.h:492:32: error: invalid type in declaration before ‘<’ token
[... 515 lines cut ...]
/usr/local/include/fst/bi-table.h: In instantiation of ‘I fst::CompactHashBiTable<I, T, H, E, HS>::FindId(const T&, bool) [with I = int; T = fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >; H = fst::ComposeHash<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >; E = std::equal_to<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >; fst::HSType HS = (fst::HSType)1u]’:
/usr/local/include/fst/state-table.h:111:67:   required from ‘fst::CompactHashStateTable<T, H>::StateId fst::CompactHashStateTable<T, H>::FindState(const StateTuple&) [with T = fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >; H = fst::ComposeHash<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >; fst::CompactHashStateTable<T, H>::StateId = int; fst::CompactHashStateTable<T, H>::StateTuple = fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >]’
/usr/local/include/fst/compose.h:409:41:   required from ‘fst::ComposeFstImpl<C, F, T>::StateId fst::ComposeFstImpl<C, F, T>::ComputeStart() [with C = fst::DefaultCacheStore<fst::ArcTpl<fst::TropicalWeightTpl<float> > >; F = fst::MatchComposeFilter<fst::Matcher<fst::Fst<fst::ArcTpl<fst::TropicalWeightTpl<float> > > >, fst::Matcher<fst::Fst<fst::ArcTpl<fst::TropicalWeightTpl<float> > > > >; T = fst::GenericComposeStateTable<fst::ArcTpl<fst::TropicalWeightTpl<float> >, fst::IntegerFilterState<signed char>, fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >, fst::CompactHashStateTable<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >, fst::ComposeHash<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > > > >; fst::ComposeFstImpl<C, F, T>::StateId = int]’
Phonetisaurus.cpp:347:1:   required from here
/usr/local/include/fst/bi-table.h:175:41: error: invalid use of incomplete type ‘struct fst::HashSet<int, fst::CompactHashBiTable<int, fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >, fst::ComposeHash<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, std::equal_to<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, (fst::HSType)1u>::HashFunc, fst::CompactHashBiTable<int, fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >, fst::ComposeHash<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, std::equal_to<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, (fst::HSType)1u>::HashEqual, (fst::HSType)1u>’
     typename KeyHashSet::const_iterator it = keys_.find(kCurrentKey);
                                         ^
/usr/local/include/fst/bi-table.h:125:9: error: declaration of ‘struct fst::HashSet<int, fst::CompactHashBiTable<int, fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >, fst::ComposeHash<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, std::equal_to<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, (fst::HSType)1u>::HashFunc, fst::CompactHashBiTable<int, fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >, fst::ComposeHash<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, std::equal_to<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, (fst::HSType)1u>::HashEqual, (fst::HSType)1u>’
 struct  HashSet : public unordered_set<K, H, E, PoolAllocator<K> > {
         ^
[... cut ...] <int, fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> >, fst::ComposeHash<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, std::equal_to<fst::DefaultComposeStateTuple<int, fst::IntegerFilterState<signed char> > >, (fst::HSType)1u>::HashEqual, (fst::HSType)1u>’
 struct  HashSet : public unordered_set<K, H, E, PoolAllocator<K> > {
         ^
Makefile:51: recipe for target 'Phonetisaurus.o' failed
make: *** [Phonetisaurus.o] Error 1`
请问如何解决这个问题？谢谢

mindcont, 2018-02-02:
通过在`is2013-conversion/phonetisaurus/src/Makefile` Line 42 中 指定使用`c++11`编译标准，可以编译通过。 
```java
THIRD_PARTIES_INCLUDE ?= -I3rdparty/sparsehash -I3rdparty/utfcpp
# EXTRA=$(CFLAGS) $(THIRD_PARTIES_INCLUDE) $(CXXFLAGS) $(CPPFLAGS)
EXTRA=$(CFLAGS) $(THIRD_PARTIES_INCLUDE) -std=c++11 -O3 $(CPPFLAGS)

POBJS=Phonetisaurus.o FstPathFinder.o MBRDecoder.o LatticePruner.o util.o
MOBJS=M2MFstAligner.o FstPathFinder.o MBRDecoder.o LatticePruner.o util.o
OOBJS=PhonetisaurusOmega.o PhonetisaurusE2F.o util.o FstPathFinder.o LatticePruner.o MBRDecoder.o
```

已有类似的issues ，参见 [issues 107 ](https://github.com/wzpan/dingdang-robot/issues/107)和  [issues 83](https://github.com/wzpan/dingdang-robot/issues/83).

andy2080, 2018-02-02:
收到，已经在编译了，非常感谢！

closed 2018-02-02 by andy2080