wzpan/dingdang-robot #119: 运行后出现我能为你做什么声音后程序结束

andy2080, 2018-02-09:
`Expression 'ioctl( devHandle, SNDCTL_DSP_CHANNELS, &numChannels )' failed in 'src/hostapi/oss/pa_unix_oss.c', line: 414
Traceback (most recent call last):
  File "dingdang.py", line 178, in <module>
    app.run()
  File "dingdang.py", line 124, in run
    self.mic.say(salutation, cache=True)
  File "/root/dingdang/client/mic.py", line 336, in say
    self.speaker.say(phrase, cache)
  File "/root/dingdang/client/tts.py", line 140, in say
    os.rename(tmpfile, cache_file_path)
OSError: [Errno 18] Invalid cross-device link`

olive900, 2018-02-10:
这个再检查一下所有的配置，看下有没有重复运行，以root身份运行等情况。

olive900, 2018-02-10:
ps auwx | grep dingdang  # 查看dingdang的PID号
kill -9 PID号

andy2080, 2018-02-11:
是root身份运行的，配置都没有问题，按照教程来配置的

wzpan, 2018-02-11:
@andy2080 试试执行如下命令：

``` sh
sudo -i # 进入 root 用户
ln -s /home/pi/dingdang /root/
ln -s /home/pi/.dingdang /root/
ln -s /home/pi/.asoundrc /root/
```

尤其是最后一句，解决麦克风问题。