wzpan/dingdang-robot #120: 网易云音乐两个问题

olive900, 2018-02-10:
第一歌单、榜单只能放第一首；第二语音命令退出不成功，然后对任何语音无响应。
还有本地music目录有音乐怎么放不出来，顺序、歌单、榜单都不会放本地音乐。
以上都在pi用户下。root用户自启动尝试多种方法均不成功。

wzpan, 2018-02-10:
更新了dingdang-contrib没？

wzpan, 2018-02-11:
@olive900 

root 用户试试执行如下命令：

``` sh
sudo -i # 进入 root 用户
ln -s /home/pi/dingdang /root/
ln -s /home/pi/.dingdang /root/
ln -s /home/pi/.asoundrc /root/
```

wzpan, 2018-02-11:
另外再更新下 musicboxapi 

``` sh
sudo pip install --upgrade MusicBoxApi
```

olive900, 2018-02-11:
以上都试过，root用户可以运行了，播放歌单出错。
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/home/pi/.dingdang/contrib/NetEaseMusic.py", line 497, in run
    self.play()
  File "/home/pi/.dingdang/contrib/NetEaseMusic.py", line 507, in play
    song = self.playlist[self.idx]
IndexError: list index out of range

olive900, 2018-02-12:
跟requirements.txt文件有关吗？
MusicBoxApi>=1.0.0
paho-mqtt>=1.3.0
eyeD3>=0.8

olive900, 2018-02-12:
今天抓到出错信息
Socket error Event: 32 Error: 10053.
Connection closing...Socket close.
Connection closed by foreign host.
Disconnected from remote host(shu) at 21:46:12.

datamonday, 2018-02-13:
遇到了同样的问题 还会卡死

datamonday, 2018-02-13:
解决了 需要把默认播放本地音乐的选项 设置为false！真的是秒播！作者很给力哈哈哈哈

olive900, 2018-02-13:
真的就这样解决了，感谢 @wzpan ，感谢 @datamonday 。

closed 2018-02-13 by olive900