wzpan/dingdang-robot #121: 语音无法唤醒叮当是什么原因？

datamonday, 2018-02-10:
对麦克风喊叮当，无法唤醒，也没有反应。用微信中的文本可以唤醒。这是什么原因？ 麦克风是respeaker 4麦的

wzpan, 2018-02-10:
驱动没装好。详见wiki安装教程，arecord，aplay可录可播再使用叮当。

datamonday, 2018-02-12:
好的好的

zhangwei188393, 2018-02-15:
镜像安装。啥都没有修改，就增加了asoundrc配置

arecord，aplay可录可播. 用的是摄像头带的mic

**** List of CAPTURE Hardware Devices ****
card 1: Camera [DVC FINE PC Camera], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~/dingdang $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

.asoundrc配置
pcm.!default {
        type asym
            playback.pcm {
                type plug
                slave.pcm "hw:0,0"
            }
            capture.pcm {
                type plug
                slave.pcm "hw:1,0"
            }        
}

ctl.!default {
        type hw
        card 0
}

启动后：

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
INFO:client.stt:PocketSphinx 识别到了：[None]
Traceback (most recent call last):
  File "dingdang.py", line 190, in <module>
    app.run()
  File "dingdang.py", line 156, in run
    conversation.handleForever()
  File "/home/pi/dingdang/client/conversation.py", line 63, in handleForever
    threshold, transcribed = self.mic.passiveListen(self.persona)
  File "/home/pi/dingdang/client/mic.py", line 244, in passiveListen
    if any(PERSONA in phrase for phrase in transcribed):
  File "/home/pi/dingdang/client/mic.py", line 244, in <genexpr>
    if any(PERSONA in phrase for phrase in transcribed):
TypeError: argument of type 'NoneType' is not iterable

是无法识别吗？没法唤醒

wzpan, 2018-02-15:
@zhangwei188393 你的代码应该不是最新的，最新的代码对 transcribed 做了个判空（见[这个提交](https://github.com/wzpan/dingdang-robot/commit/825163c1ce47cd9d053685abbbb61bc064dfaac3)），是不会导致这个错误的。

zhangwei188393, 2018-02-16:
git pull 更新最新之后
如需查看log，可以执行 `tail -f 叮当所在目录/temp/dingdang.log`

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Traceback (most recent call last):
  File "dingdang.py", line 178, in <module>
    app.run()
  File "dingdang.py", line 125, in run
    conversation.handleForever()
  File "/home/pi/dingdang/client/conversation.py", line 67, in handleForever
    threshold, transcribed = self.mic.passiveListen(self.persona)
  File "/home/pi/dingdang/client/mic.py", line 238, in passiveListen
    any(PERSONA in phrase for phrase in transcribed):
  File "/home/pi/dingdang/client/mic.py", line 238, in <genexpr>
    any(PERSONA in phrase for phrase in transcribed):
TypeError: argument of type 'NoneType' is not iterable

wzpan, 2018-02-16:
@zhangwei188393 能不能把你client/mic.py里第238行附近的内容发一下？

wsyccc, 2018-02-16:
http://proxy.peiyinge.com:17063/synth?ts= 这个API连接 好像失效了吧

zhangwei188393, 2018-02-22:
我代码没有动过， 直接Git pull了。

wzpan, 2018-02-22:
@zhangwei188393 我需要看代码，确认下你的代码是否最新。

wzpan, 2018-02-24:
没有收到进一步反馈，先关闭本issue。等后续有反馈我再重新开启跟进。

closed 2018-02-24 by wzpan