wzpan/dingdang-robot #125: 唤醒词没有激活成功

HuPengCheng, 2018-02-28:
叫叮当，日志中明显已经接收到了，但是没有提示我能为您做什么，也没有蜂鸣声
2018-02-28 22:52:39,776 tts.py[line:130]         INFO found speech in cache, playing...[/usr/workspace/dingdang/temp/ali-tts6bdcbf7fd6c8e8a66e67c78aea445e32.mp3]
2018-02-28 22:52:43,900 conversation.py[line:49]         INFO Starting to handle conversation with keyword '嘿叮当'.
2018-02-28 22:52:46,650 stt.py[line:220]         INFO PocketSphinx 识别到了：['DINGDANG']
2018-02-28 22:52:46,652 conversation.py[line:72]         INFO Nothing has been said or transcribed.
2018-02-28 22:52:57,219 conversation.py[line:72]         INFO Nothing has been said or transcribed.
2018-02-28 22:52:59,834 stt.py[line:220]         INFO PocketSphinx 识别到了：['DINGDANG']
2018-02-28 22:52:59,835 conversation.py[line:72]         INFO Nothing has been said or transcribed.
2018-02-28 22:53:01,856 stt.py[line:220]         INFO PocketSphinx 识别到了：['']
2018-02-28 22:53:01,857 conversation.py[line:72]         INFO Nothing has been said or transcribed.
2018-02-28 22:53:12,424 conversation.py[line:72]         INFO Nothing has been said or transcribed.
2018-02-28 22:53:22,992 conversation.py[line:72]         INFO Nothing has been said or transcribed.
2018-02-28 22:53:33,560 conversation.py[line:72]         INFO Nothing has been said or transcribed.
2018-02-28 22:53:44,128 conversation.py[line:72]         INFO Nothing has been said or transcribed.
2018-02-28 22:53:54,696 conversation.py[line:72]         INFO Nothing has been said or transcribed.

wzpan, 2018-02-28:
keyword 是嘿叮当，不是叮当。所以你要说的是“嘿叮当“，识别到dingdang是没有用的。

HuPengCheng, 2018-02-28:
也就是说，唤醒词词典文件里面要写“嘿叮当”是吗，默认目前只有DINGDANG和XIAOMEI

wzpan, 2018-03-01:
手动安装用到的 vocabularies 有问题，我刚刚才发现。已经重新制作上传，请下载新版本的 vocabularies.zip，然后使用唤醒词 “嘿叮当” 唤醒叮当。

HuPengCheng, 2018-03-01:
我看到你上传的唤醒词字典里面写的全拼，麻烦提供中文声学模型生成的方法。
另外，profile.yml中robot_name是否与声学模型字典里面填写的要求完全一致，还是像你新提交的那样全拼也可以。
还有就是profile.yml中robot_name是否可以设置多个（如果可以怎么设置）。

wzpan, 2018-03-01:
更改唤醒词见：https://github.com/wzpan/dingdang-robot/wiki/configuration#%E4%BF%AE%E6%94%B9%E5%94%A4%E9%86%92%E8%AF%8D
其实并没有针对中文全拼，只是挑选了一个中英文念法相似的唤醒词。

HuPengCheng, 2018-03-01:
profile.yml中robot_name和dictionary里面的词没有必须一致的要求是吗

wzpan, 2018-03-01:
必须一致

closed 2018-03-10 by wzpan