wzpan/dingdang-robot #126: 运行时出现了无法从tzlocal 导入 get_localzone

Durantdzr, 2018-03-02:
启动后报错：
Traceback (most recent call last):
  File "dingdang.py", line 16, in <module>
    from client.conversation import Conversation
  File "/home/pi/dingdang/client/conversation.py", line 4, in <module>
    from .notifier import Notifier
  File "/home/pi/dingdang/client/notifier.py", line 6, in <module>
    from apscheduler.schedulers.background import BackgroundScheduler
  File "/usr/local/lib/python2.7/dist-packages/apscheduler/schedulers/background.py", line 4, in <module>
    from apscheduler.schedulers.base import BaseScheduler
  File "/usr/local/lib/python2.7/dist-packages/apscheduler/schedulers/base.py", line 10, in <module>
    from tzlocal import get_localzone
ImportError: cannot import name get_localzone
我用pip list 检查出tzlocal是1.5.1版本，也重新安装过，没法运行，求助。。

wzpan, 2018-03-03:
试试更新一下 apscheduler ？

Durantdzr, 2018-03-03:
嗯，我现在已经解决了，卸载了1.5.1版本，安装1.1.1的旧版本。
应该是tzlocal-1.5.1里面对于get——localzone有了更改吧
ww

closed 2018-03-10 by wzpan