wzpan/dingdang-robot #127: 运行后无反应直接弹出到命令行，请问是怎么回事

wangshuai1320, 2018-03-03:
python dingdang.py后出现了那个信息框框，但是信息框框之后只出现了一句log相关的提示后就直接退出到命令行嘞。
用的是2017-10-29的镜像，然后按照配置慢慢配置的，请问是怎么回事呢

wzpan, 2018-03-03:
看下真正的log：

``` sh
cat dingdang/temp/dingdang.log
```

wangshuai1320, 2018-03-03:
您好，感谢您的回复。
log上显示的是
File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 127, in c                            ompose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 98, in chec                            k_event
    self.current_event = self.state()
  File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 439, in par                            se_block_mapping_key
    "expected <block end>, but found %r" % token.id, token.start_mark)
ParserError:while parsing a block mapping
in "/home/pi/.dingdang/profile.yml", line 1, column 1
expected <block end>, but found '<block mapping start>'
  in "/home/pi/.dingdang/profile.yml", line 73, column 5

wzpan, 2018-03-03:
应该是 yaml 格式错误，检查下你的 profile.yml 配置哈。

wangshuai1320, 2018-03-04:
您好，我的问题解决了，确实是yaml的格式错误。
dingdang跑起来之后，无法识别语音，查看log都是空的，我看了您的 视频，是Mic的问题么？但是我录音录到的声音还是比较大的，只是有很多滋滋的噪声。
请问我是需要换个Mic么？买那个ReSpeaker就可以么？
2018-03-04 11:45:47,186 stt.py[line:347]             INFO 百度语音识别到了:
2018-03-04 11:45:47,188 mic.py[line:326]             INFO 机器人说：什么?
2018-03-04 11:45:51,635 stt.py[line:220]             INFO PocketSphinx 识别到了                                                    ：['']
2018-03-04 11:45:51,637 conversation.py[line:72]             INFO Nothing has be                                                    en said or transcribed.
2018-03-04 11:45:54,653 stt.py[line:220]             INFO PocketSphinx 识别到了                                                    ：['']
2018-03-04 11:45:54,654 conversation.py[line:72]             INFO Nothing has be                                                    en said or transcribed.
2018-03-04 11:45:56,851 stt.py[line:220]             INFO PocketSphinx 识别到了                                                    ：['']
2018-03-04 11:45:56,852 conversation.py[line:72]             INFO Nothing has be

wzpan, 2018-03-10:
Mic确实不行，考虑换一个

closed 2018-03-17 by wzpan