wzpan/dingdang-robot #132: 启动报错

JasonIonke, 2018-03-12:
运行python dingdang.py显示如下，麻烦wzpan帮忙看下，谢谢！

/usr/local/lib/python2.7/dist-packages/requests/__init__.py:80: RequestsDependencyWarning: urllib3 (1.22) or chardet (2.3.0) doesn't match a supported version!
  RequestsDependencyWarning)

*******************************************************"
*             叮当 - 中文语音对话机器人               *
*          (c) 2017 潘伟洲 <m@hahack.com>             *
*   https://github.com/wzpan/dingdang-robot.git       *
*******************************************************

如需查看log，可以执行 `tail -f 叮当所在目录/temp/dingdang.log`

Traceback (most recent call last):
  File "dingdang.py", line 154, in <module>
    level=logging.INFO)
  File "/usr/lib/python2.7/logging/__init__.py", line 1540, in basicConfig
    hdlr = FileHandler(filename, mode)
  File "/usr/lib/python2.7/logging/__init__.py", line 911, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python2.7/logging/__init__.py", line 936, in _open
    stream = open(self.baseFilename, self.mode)
IOError: [Errno 2] No such file or directory: '/home/pi/Desktop/dingdang/temp/dingdang.log'

licshire, 2018-03-12:
手动创建temp目录：
1、切换到pi用户
sudo su - pi
2、新建temp目录
mkdir -p /home/pi/Desktop/dingdang/temp/

JasonIonke, 2018-03-12:
谢谢licshire！ 问题解决了

closed 2018-03-12 by wzpan