wzpan/dingdang-robot #135: 手动安装 都完成安装了但是OSError: libpulse.so.0: cannot open shared object file: No such file or directory怎么回事

mapenghui03, 2018-03-14:
No handlers could be found for logger "cmuclmtk"
/usr/local/lib/python2.7/dist-packages/pydub/utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
  warn("Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work", RuntimeWarning)
Traceback (most recent call last):
  File "dingdang.py", line 37, in <module>
    from client.mic import Mic
  File "/home/pi/dingdang/client/mic.py", line 16, in <module>
    import pulsectl
  File "/usr/local/lib/python2.7/dist-packages/pulsectl/__init__.py", line 4, in <module>
    from . import _pulsectl
  File "/usr/local/lib/python2.7/dist-packages/pulsectl/_pulsectl.py", line 627, in <module>
    pa = LibPulse()
  File "/usr/local/lib/python2.7/dist-packages/pulsectl/_pulsectl.py", line 579, in __init__
    p = CDLL('libpulse.so.0')
  File "/usr/lib/python2.7/ctypes/__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libpulse.so.0: cannot open shared object file: No such file or directory

wzpan, 2018-03-14:
再更新下代码，应该fix了。

mapenghui03, 2018-03-14:
我是小白该 怎么操作呢

wzpan, 2018-03-15:
cd dingdang
git pull

closed 2018-03-15 by wzpan