wzpan/dingdang-robot #136: 运行以后唤醒叮当后回答“什么”就没反应了怎么回事

mapenghui03, 2018-03-14:
运行以后唤醒叮当后回答“什么”就没反应了怎么回事/usr/local/lib/python2.7/dist-packages/pydub/utils.py:165: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
  warn("Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work", RuntimeWarning)
还有这样的警告怎么解决

mapenghui03, 2018-03-14:
警告解决了可是还是没有办法 正常使用只是 说“什么” 

*******************************************************"
*             叮当 - 中文语音对话机器人               *
*          (c) 2017 潘伟洲 <m@hahack.com>             *
*   https://github.com/wzpan/dingdang-robot.git       *
*******************************************************

如需查看log，可以执行 `tail -f 叮当所在目录/temp/dingdang.log`

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
No matches.

也只有这样说话也没有反应

wzpan, 2018-03-15:
先看下常见问题 “叮当总是过快结束主动聆听” 部分：https://github.com/wzpan/dingdang-robot/wiki/troubleshooting#%E4%BD%BF%E7%94%A8%E9%97%AE%E9%A2%98

closed 2018-03-18 by wzpan