wzpan/dingdang-robot #138: 指令无法识别

wzr12345, 2018-03-17:
[image]
启动后能听到机器人说我能为您做什么,接下来说了也没有反应,然后我就按下ctrl+z.
我发现traceboak跟troubleshooting中的启动后报错,相似,可是dingdong的进程终止了,还是运行不了

wzpan, 2018-03-17:
arecord，aplay 能否正常工作？

wzr12345, 2018-03-17:
yes, ceshi meiyou wenti.
nengluyin ,nengting.

-----------------------------------

wzpan, 2018-03-17:
要说唤醒词“嘿叮当”才能唤醒叮当

wzr12345, 2018-03-17:
是的，他能识别到 [heidingdang],随即我说 唱歌，然后就显示 "info nothinghas been said or transcribed" 循环。先睡了，晚安

wzpan, 2018-03-18:
识别到会响滴一声，滴一声了再说话

wzr12345, 2018-03-18:
成功了,问题出现在我修改了
robot_name: 'DINGDANG'  # 必须使用大写
robot_name_cn: '叮当'
    起先,我看到提醒词是"heydingdang",与这个robot_name是"dingdang",没有关系,所以我就改变了robot_name.
自己理解错误
感谢您的指导
♪(＾∀＾●)ﾉ

closed 2018-03-18 by wzr12345